ms-iot/ntvsiot #23: Deployment of Node Express UWP application to x86 fails

mikehall-ms, 2015-07-06:
installing NTVS from \\scratch2\scratch\munyirik\Node.js\BundleInstaller - generating a default Node Express UWP application, setting build to x86, configuring project properties for x86 device IP address, building and deploying fails with error message DEP0700: Registration of the app failed. Windows cannot install package Express4UWPApp6 because the package requires Architecture ARM, but this computer has architecture x86 (0x80073d10).

the project bin\appmanifest.xml has line 10 <Identity Name="Express4UWPApp6" Publisher="CN-mikehall" version="1.0.0.0" ProcessorArchitecture="arm" />

munyirik, 2015-07-07:
Fix will be in the next release

munyirik, 2015-07-27:
As a workaround for now, rebuilding (Build Menu -> Rebuild Solution) is required if architecture is changed.

commit 4e0a9045 (2016-01-26): fix #23
closed 2016-01-26 by munyirik