ms-iot/ntvsiot #25: Error : socket hang up error

Dharshz, 2015-07-07:
When trying to install typed definition files through the tsd. I am opening current directory through cmd and run "tsd install express"

munyirik, 2015-07-07:
Hi @Dharshz, I'm not able to reproduce this issue. It sounds like a problem that's independent of NTVS since you're just running the tsd command (i.e. you would see the error in a different directory as well). If you provide me with detailed steps to reproduce the issue I can investigate more.

Dharshz, 2015-07-08:
@munyirik I think this issue comes with the Network issue. I ran tsd install express within office network then I faced this issue but I tried with my home machine it worked fine. I suspect there is something related to network proxy. Thank you for your response

closed 2015-07-08 by Dharshz