ms-iot/ntvsiot #27: Can't build Node.js apps on VS 2015 with Windows 10 tools 23116.00

BoolLi, 2015-07-24:
Hello,

I'm using  VS 2015 RTM with Windows 10 tools 23116.00, and I followed the instructions here:
http://ms-iot.github.io/content/en-US/win10/samples/NodejsWU.htm
However, when I tried to build and run the app, I got the error message below:

    Error		Could not find SDK "Microsoft.VCLibs.AppLocal, Version=14.0".	NodejsWebServer3	C:\Program Files (x86)\MSBuild\14.0\bin\Microsoft.Common.CurrentVersion.targets	2048

I might need to add a reference to the SDK but I'm not sure how. 

Any help will be appreciated! Thanks!

munyirik, 2015-07-24:
Hello @BoolLi 
A new release will be available very soon that works with the RTM build. To get unblocked now, try the following:
1. Replace c:\program files (x86)\MSBuild\Microsoft\visualstudio\v14.0\node.js tools\Microsoft.nodejsuwp.targets with https://github.com/ms-iot/ntvsiot/blob/master/Setup/NodejsUwpFiles/Microsoft.NodejsUwp.targets
2. Add the following properties to your njsproj file -
    <WindowsTargetPlatformVersion>10.0.10069.0</WindowsTargetPlatformVersion>
    <WindowsTargetPlatformMinVersion>10.0.10069.0</WindowsTargetPlatformMinVersion>

Note: Make sure that the version number in the property above corresponds with "C:\Program Files (x86)\Windows Kits\10\Include\<newest version number>"

BoolLi, 2015-07-24:
It works now! Thank you!

closed 2015-07-24 by BoolLi