ms-iot/ntvsiot #34: NodeJS Express UWP build fails when utilizing Bower Packages

daveraskin, 2015-08-18:
Fails with "Payload contains two or more files with the same destination path".

This makes use of bower almost impossible since many components include an "index.js", "app.js" or some other such thing.

munyirik, 2015-08-18:
Hi @daveraskin, I wasn't able to reproduce this problem when I tried it. I have the latest tools installed, created a web server project, used the UI to install bower, and built successfully. Could you provide me the exact steps to reproduce if possible? 

Another problem you will run into even though you are able to build is failure to deploy. The problem is that you have no control over where the Node.js UWP app is installed on the device. So the node_module dependency tree will easily hit MAX_PATH (this manifests as a “cannot copy file X..” in the VS output window). To get around this you can upgrade npm to 3.2.2 (I believe this is the latest). To do this:
1.	Run ‘npm install -g npm-windows-upgrade’
2.	Then run ‘npm-windows-upgrade --version:3.2.2’
After this, is done THEN install bower. Note that npm v3 isn’t supported in NTVS yet but you should still be able to build and deploy.

daveraskin, 2015-08-18:
@munyirik I created the project, installed the bower projects (from the powershell command line), included them in the project (because I was not able to utilize them otherwise), and built.  I wasn't aware there was a UI for installing bower components, so that may be my issue.  I am using node tools v 1.1 and have just upgraded to npm 3.2.2.

munyirik, 2015-08-18:
@daveraskin to do it through the UI, right click on the npm node in the solution explorer then select "Install new npm packages..." - > you can search for bower and then install.

daveraskin, 2015-08-18:
@munyirik The issue isn't actually with bower itself, but with the power packages once they're in my solution.  Bower installs the packages correctly, but once they're part of my solution, building throws the payload error.

daveraskin, 2015-08-19:
bower packages*

munyirik, 2015-08-19:
are the packages in the node_modules folder or the root where the project file is?

daveraskin, 2015-08-19:
@munyirik In the "public" folder one level below the root level, where the front end can access them.

munyirik, 2015-08-19:
I think that's the problem. The project system only knows how to copy npm package files that are in the node_modules folder. Is it possible to move the public folder?

daveraskin, 2015-08-20:
@munyirik it is not possible to move them into the node_modules folder or outside of the project folder.

munyirik, 2015-08-20:
@daveraskin To get unblocked try this workaround (it's basically how node_modules gets copied).
Open the following file as an admin: C:\Program Files (x86)\MSBuild\Microsoft\VisualStudio\v14.0\Node.js Tools\Microsoft.NodejsUwp.targets

Add the following ItemGroup:

  <ItemGroup>
    <PublicDir Include="$(ProjectHome)\public"/>
  </ItemGroup>

Then in the CustomOutputGroupForPackaging target, add the following:

    <AssignTargetPath Files="@(PublicDir->'%(FullPath)')" RootFolder="@(PublicDir)" Condition="Exists(@(PublicDir))">
      <Output TaskParameter="AssignedFiles" ItemName="CustomOutputGroupForPackagingOutput" />
    </AssignTargetPath>

Close, reopen you project and try and build again.

daveraskin, 2015-08-20:
@munyirik Thanks.  Is there a block of code missing after you say "add the following ItemGroup"?

daveraskin, 2015-08-20:
@munyirik It worked!  Thanks very much.  Since we're pushing express servers pretty hard on windowsondevices.com, it may behoove us to have something like this in place by default, since we claim to have support for bower and npm.  Since bower is mostly client-side packages, they will almost always be located in express's default "public folder".  Thanks again!

munyirik, 2015-08-20:
@daveraskin great! So there's actually an [Express template](http://ms-iot.github.io/content/en-US/win10/samples/NodejsWUExpress.htm) we provide that does create a public folder. It only has one file (public\stylesheets\style.styl). I'm curious how you put together your project (assuming you didn't use the template). What files do you have in your public folder?

daveraskin, 2015-08-20:
@munyirik I used the express template. the public folder is where all the server-side files and directories that the client will need to access (stylesheets, javascript files, etc.) which is why that's where my "bower_components" folder needs to be.  So, for people to be able to effectively utilize bower, they'll need the code you gave me in their Microsoft.NodejsUwp.targets file.

daveraskin, 2015-08-20:
where all the server-side files and directories that the client will need to access (stylesheets, javascript files, etc.) *are located* *

munyirik, 2015-08-20:
@daveraskin got it. thanks. Then let's leave this issue open and I'll fix asap.

munyirik, 2015-09-04:
@daveraskin the fix to include public folder been added to the latest release - https://github.com/ms-iot/ntvsiot/releases/tag/v1.2.

closed 2015-12-04 by ooeygui