ms-iot/ntvsiot #47: The activation request failed with error 'There are no more endpoints available from the endpoint mapper'.

mikeptweet, 2015-09-18:
Tried running the default NodeJs Web Server project on a fresh WIndows 10, VS 2015 machine (x64)

Can I Debug/Run the app on buld machine?

---------------------------
Microsoft Visual Studio
---------------------------
Unable to activate Windows Store app 'NodejsWebServer2_dg3rhm3j3sdc0!App'. The activation request failed with error 'There are no more endpoints available from the endpoint mapper'.

See help for advice on troubleshooting the issue.
---------------------------
OK   Help   
---------------------------

munyirik, 2015-09-18:
Hi @mikeptweet 
We currently don't support debugging or running the app on the build machine. Lets keep this issue open as I may have another update on this soon.
Do you have a Raspberry Pi 2 or MBM you can use?
Thanks.

mikeptweet, 2015-09-18:
OK ... thanks, I'm setting up a Raspberry Pi 2 to test

jaspreetd, 2015-10-26:
I am facing same deploying on emulator as well as local and also trying to deploy to Raspberry PI 2. I am getting this same error in all cases. I even tried turning off the firewall on my machine as read in other forums related to this error.

Please let me know if anyone has found any solution to this issue.

jaspreetd, 2015-10-26:
[image]

munyirik, 2015-10-26:
Hi @jaspreetd, We'll be publishing instructions on how to deploy to your desktop machine very soon as part of the next Windows 10 IoT Core release.

jaspreetd, 2015-10-26:
Thanks Munyiri, looking forward for the fix and instructions.

Jaspreet

andreabeggiato-geekcups, 2015-11-07:
Hi guys,
I'm facing the same issue trying to debug an IBackgroundTask. I need to deploy it to a board?

bfjelds, 2015-11-07:
Currently, Background Application (IoT) and Node.JS Web Server requires deployment to a device.

munyirik, 2015-12-04:
Hello Everyone,
You can now deploy to Windows 10 desktop by allowing "Embedded Mode" 
For instructions on how to do that please follow the instructions on this page: http://ms-iot.github.io/content/en-US/win10/EmbeddedMode.htm
Thanks.

ZahoorBaloch, 2015-12-07:
Hello, I am trying to run Headless project of Alljoyn DSB on local machine, I have followed the given instruction but still I am having below error. 

[image]

munyirik, 2015-12-07:
Hi @ZahoorBaloch could you open Task Manager, go to the Services tab and check that "embeddedmode" service is running?

ZahoorBaloch, 2015-12-08:
No, it's not running, It has starting status.
[image]

paulmon, 2015-12-08:
This looks like a problem that was fixed in the Windows 10 Fall Update (build 10586.)

munyirik, 2015-12-08:
Thanks @paulmon. 
@ZahoorBaloch you can get more details about updating your machine with this link - http://windows.microsoft.com/en-us/windows-10/windows-update-faq

andreabeggiato-geekcups, 2015-12-08:
Hi,
first of all thanks for your works! Now I can debug a nodejs application, but if it is a web server application I can't reach it with my browser. If I start the debug and type in the browser http://localhost:port I didn't get any response. My firewall is turned off, but I think I'm missing something.

Bye!

munyirik, 2015-12-09:
Hi @andreabeggiato-geekcups, I *think* there's a restriction that blocks desktop apps (i.e. your browser) from accessing a server hosted by a UWP application (i.e. the Node.js UWP application). You should be able to access the server from a different machine or from a UWP application. I'll look into whether there's a workaround for you to use your browser and get back to you.

andreabeggiato-geekcups, 2015-12-09:
Thanks @munyirik, I can confirm that from another machine in the same network (I tried with a VM) it's work like a charm!

ZahoorBaloch, 2015-12-09:
Thanks Guys! I will update my machine and provide you feedback ASAP, Cheers.

munyirik, 2016-01-22:
Closing this now since embedded mode will solve this issue. If any other issues are seen please feel free to open a new issue. Thanks!

closed 2016-01-22 by munyirik