ms-iot/ntvsiot #52: Express + Stylus : compiling CSS doesn't work

kurolounet, 2015-10-22:
Hello,

I create a project Express 4 (with Jade and Stylus for templage engine and preprocessor CSS).
But when I launch the debug (or deploy), .styl files don't want to compile in .css files.
I don't understand why. No errors.
Someone can help me, please ? Thank's.
 
I'm using Express 4, Jade v1.6.0 and Stylus v0.52.4.
 
This is my app.js code : 

var express = require('express');
var path = require('path');
var favicon = require('serve-favicon');
var logger = require('morgan');
var cookieParser = require('cookie-parser');
var bodyParser = require('body-parser');

var routes = require('./routes/index');
var users = require('./routes/users');

var app = express();

// view engine setup
app.set('views', path.join(__dirname, 'views'));
app.set('view engine', 'jade');

// uncomment after placing your favicon in /public
//app.use(favicon(__dirname + '/public/favicon.ico'));
app.use(logger('dev'));
app.use(bodyParser.json());
app.use(bodyParser.urlencoded({ extended: false }));
app.use(cookieParser());
app.use(require('stylus').middleware(path.join(__dirname, 'public')));
app.use(express.static(path.join(__dirname, 'public')));

munyirik, 2015-10-22:
Hello @kurolounet,
I think what's happening is that the app cannot write (due to UWP app container restrictions) to the stylesheets directory. So the file is actually getting created in the local storage folder of the application (C:\Users\DefaultAccount\AppData\Local\Packages\...\LocalState\). It's not ideal (i.e. a bug) and what you should see in an error. I'll look into this but in the meantime, here are some options to work around this issue:
1. Compile the css manually and then include it in your project.
2. I haven't tried this yet but there's a tool in https://github.com/ms-iot/iot-utilities that you could use to give your app permission to write to that folder (C:\Users\DefaultAccount\AppData\Local\DevelopmentFiles\Your app\...\StyleSheets).

kurolounet, 2015-10-23:
Thx a lot.
I try to create the .css file and it works.

I didn't try the tool that you say but I will do later.

Thx again !

munyirik, 2015-10-23:
Hi @kurolounet, how did you try to compile it?

kurolounet, 2015-10-24:
I run the debug mode in my Raspberry Pi.
After that, I connect on it with 192.168.0.X.
I see the compiling css file in the LocalState but not in my project.

Do you need more information ?

munyirik, 2015-10-24:
Ok, easiest way to get unblocked is to add the file is to copy it from the device to the public\stylesheets folder in your project on your PC. Then in the project Solution Explorer, right click on the css file and select Include in Project. We'll fix this issue in a future release so keep this issue open. Thanks!!

kurolounet, 2015-10-24:
I've already done that. The only solution is to copy the file from the device (LocalState) to the desktop project on the PC and after, deploy it (or debug it). It's a little complicated but it's the only solution.

I'm waiting a future release to fix the issue. Thank's to you !

munyirik, 2015-11-30:
This is fixed in https://github.com/ms-iot/ntvsiot/releases/tag/v1.3

munyirik, 2015-12-04:
@kurolounet Have you had a chance to trying out the latest release with the fix? Let me know so we can close this issue.
Thanks!

kurolounet, 2015-12-05:
Yes, but it can't work for this moment. I can't launch the debug mode because there are errors. I think it's because my Windows IoT isn't updated.
I will update this and retry.

kurolounet, 2015-12-07:
I retry and it still doesn't work.
After some research, I found an article on Microsoft.com about that : VS2015 debugging mode can't run on the latest version of Windows IoT 10586.
http://social.technet.microsoft.com/wiki/contents/articles/32718.getting-visual-studio-debugger-running-on-latest-build-10586-of-windows-10-iot.aspx

I'm trying to find the previous version (10556) to find the RDBG folder.

bfjelds, 2015-12-07:
Windows IoT 10586 does not come pre-configured with MSVSMON (the remote debugger contained in RDBG) because this build is intended to work with the latest Visual Studio update (http://blogs.msdn.com/b/visualstudio/archive/2015/11/30/visual-studio-update-1-rtm.aspx).  With this Visual Studio update, MSVSMON is pushed to the device and started as part of app deployment.  To make this happen you should only need to provide your device name (or ip address) and select Universal authentication.

bfjelds, 2015-12-07:
You can find more information about build 10586 at windowsondevices.com and about app deployment in the Hello World sample:  http://ms-iot.github.io/content/en-US/win10/samples/HelloWorld.htm.

munyirik, 2015-12-07:
Node.js projects will use Universal authentication automatically so there's no need to set that (also there's no UI to set it). But as bfjelds mentioned build 10586 of Windows 10 IoT Core does not come with MSVSMON (i.e. the RDBG folder). Visual Studio deploys it automatically. 
Is there a specific error message your seeing that you can share?

kurolounet, 2015-12-07:
Thank's a lot. It works.
I updated my VS2015 version and it works.

closed 2015-12-07 by kurolounet