ms-iot/ntvsiot #62: Error deploying windows iot app with FileTypeAssociation declaration

aronbardsley, 2016-01-02:
Hi,

Love the work you guys are doing.

I am attempting to add a file type association to my project (see below). I set all of the required properties as per the package designer and attempt to deploy my app locally but I receive the following error:
Severity	Code	Description	Project	File	Line	Suppression State
Error		Error : DEP0700 : Registration of the app failed. AppxManifest.xml(32,10): error 0x80070490: Cannot register the BackgroundApplication1-uwp_1.0.0.0_x86__fcj39y6ajd45j package because the following error was encountered while parsing the windows.fileTypeAssociation Extension element: Element not found.
.  Try again and contact the package publisher if the problem persists. (0x80073cf6)	BackgroundApplication1			

I have tried: 
- Creating a new Blank App (Universal Windows) - this works with FileTypeAssociation
- Created a new Windows IOT app - this gives me the error above.

To reproduce:
1. Create a new Windows IOT app using the template in visual studio
2. Update the package manifest file to include a file type association declaration for any file format, fill in the required fields.
3. Save the package manifest file and attempt to deploy locally

I've included a screenshot below for extra detail. 

[image]

munyirik, 2016-01-05:
Hi @aronbardsley and thanks! (sorry for the delay...back from holidays). I'll look into this issue and get back to you.

jessekaplan, 2016-01-06:
Thanks for reporting this. The error message could definitely use some work, but it's not expected that you could register a BackgroundApp as a file extension handler. These handlers are strictly UI apps and go through an activation pathway that doesn't exist for background apps. 

If you want to register a file handler in your app, you need to build a UI app to handle that and either include it in same app as the IBackgroundTask or place it on another app that gets deployed to your device. 

Out of curiousity, why do you want to register a file extension handler for your IoT device? 

Thanks,
Jesse

aronbardsley, 2016-01-06:
Hi Jesse,

I'm writing an app that:
1. takes video from IP Cameras via RTSP
2. Performs some image processing on the video feeds
3. Stores important events on an external hard drive connected to the raspberry pi
4. Serves web requests
5. Synchronizes important events up to azure

My requirements meant that i had to make this deployable to a computing device with a small footprint (physical, electrical etc.). I'd already started writing the software in C# about a year ago, knowing that with Microsoft's new focus on portability, i'd be able to figure something out.

I am fairly new with windows app development model, so I wasn't sure if my question was ignorant or not. I picked "BackgroundTask" type app, because I do not need a UI on the actual device - just a headless app.Are you saying that, to run a backgroundApp on Windows IOT - I don't need to register a file extension handler to access files on the disk or should I be using a UI app?

I attempt to run my BackgrroundApp on my desktop machine to debug, but I get an access denied exception when I attempt to access files on the local disk. Perhaps I need to test this further. 

Thanks again for your help

jessekaplan, 2016-01-07:
Ah ok. Yeah, you don't need to worry about the file association extensions for that. The file extension is basically used to specify that you want your app to be used when a user double clicks on a file to open it. You don't need to worry about that if you are just opening or saving files of a particular type. 

As far as running them on the desktop is concerned, BackgroundApps, by default, only work on IoT Core as they are allowed to use system resources in a much less restricted way that would be inappropriate for most store deployed apps. However, if you want to run these on desktop first, you can enabled "Embedded Mode" that will allow your background apps to be activated. You can see the docs here: http://ms-iot.github.io/content/en-US/win10/EmbeddedMode.htm

aronbardsley, 2016-01-08:
Hi Jesse,

I don't have my raspberry pi handy at the moment, so I have been trying to get this to work with windows embedded mode on my desktop (just for debugging purposes).

I have done further testing and found that the following code will produce an access denied error in a BackgroundApp running on the desktop in embedded mode. 

[image]

I've read the below msdn article and it is telling me that to access files on removable storage (including create - what I am trying to achieve), I need to declare a file type handler. 
https://msdn.microsoft.com/en-au/library/windows/apps/br211423.aspx

>Removable storage, such as an external hard drive or USB flash drive, or MTP portable device, including the capability to add, change, or delete specific files. This package can only access file types that it has declared in the manifest.

jessekaplan, 2016-01-11:
My apologies for the earlier incorrect answer, it does look like the File Extension Handler is required for removable storage access. We are looking into this with the owning team and may be able to address this in a future release. 

In the meantime, you can work around this by creating an empty XAML app and then adding your background tasks to the VS solution and include those background tasks in your appx. You would then register the UI app as a file extension handler, but not do anything with it. This is a bit of a pain, but shouldn't be noticeable from the end users on the IoT Core devices. Deploying the same app on desktop would result in people being able to activate your UI app, but it's likely no-one would select it and if they did you could always put a bit of text in the XAML app recommending they choose something else.  

I'll respond back to this issue with any updates about a longer term fix for this. 

Thanks,
Jesse

aronbardsley, 2016-01-11:
Hi Jesse,

No problem - I'll give the work-around a go in the next couple of days.
I look forward to any long term fix that could be provided for this

Thanks again for your help

aronbardsley, 2016-01-27:
This is is now working with a UI app that is deployed to the raspberry pi. Selecting the file type extensions worked as expected (for the most part, see below). I can now access removable storage on the Raspberry Pi.

I did have one issue with the ttf file extension (font) which I couldn't register (the app store wouldn't let me deploy my app). I changed the file to a different extension as I was serving it through my web server and the browser doesn't look at the file extension anyway (as it looks at the content type). So this isn't an issue for me.

Thanks again for the help on providing a work around.

closed 2016-01-27 by aronbardsley

playbass, 2016-07-12:
Any word from the owning team if a fix for headless apps accessing remove storage is coming in a future release?

PeterSawyer, 2016-08-05:
@jessekaplan any progress on log term fix for this?

mmccullo, 2016-10-21:
@jessekaplan consider the scenario where you have a single foreground UI app and a background task that handles low-level details. The documentation for IoT states that shared memory is not an option and that data sharing should occur via persistent storage.

Without the ability of the background task to create files, the data sharing between tasks (i.e. apps) can only be one-way -- from the UI app to the background task.

What is the workaround for the background task to post shared data if I already have a foreground UI app?

The foreground and background are not easily integrated in my case because the foreground is UI in C# and the background is Arduino-compatible C++/CX.

jessekaplan, 2016-10-21:
Still no word on a long term fix, but you have two good options here. 

I would recommend that you take advantage of the "AppServices" feature that is designed to allow different apps to easily communicate with each other. Under the covers it uses named pipes, but the API allows a very easy string based messaging system. We actually have a sample of a C# app communicating with an Arduino-compatible C++/CX app. In this sample the C# app is also a background app, but it works the same either way. You can find that sample here:  https://github.com/ms-iot/samples/tree/develop/ArduinoLibraryBlinky 

Secondly, while neither foreground nor background apps have full access to the file system, there is a way for two apps from the same publisher to easily share files between themselves. The ApplicationData class has a method "GetPublisherCacheFolder" that gives an app access to a folder that is shared between any apps with the same publisher. You can find docs for this here: 
https://msdn.microsoft.com/library/windows/apps/windows.storage.applicationdata.getpublishercachefolder.aspx

munyirik, 2016-10-21:
@mmccullo the zip file below has a sample of how appservices can be used. The example is using a node background app and a C# background app (but it would work with a foreground app too).

[app_services_sample.zip](https://github.com/ms-iot/ntvsiot/files/545175/app_services_sample.zip)

mmccullo, 2016-10-24:
@jessekaplan it appears GetPublisherCacheFolder() is not implemented on Windows 10 IoT Core. When I try to use it in a C++/CX app, I get:

StartupTask.cpp(239): error C2039: 'GetPublisherCacheFolder': is not a member of 'Windows::Storage::ApplicationData::Current'
StartupTask.cpp(239): error C3861: 'GetPublisherCacheFolder': identifier not found

.NET library items are generally not available in native IoT apps, correct? C++/CX and C++/CLI cannot be combined.

I will investigate the AppServices feature.

jessekaplan, 2016-10-24:
Thanks for letting us know about your issues with GetPublisherCacheFolder, we are looking into it. 

It is actually available on IoT Core and we've verified it for C# apps, but have repro'd the error you are seeing with C++/CX. This is actually a Windows UWP API, not a .Net one, and so should be fully supported for C++/CX and not just C#. We'll keep you appriased of our progress.

munyirik, 2016-10-25:
@mmccullo when I reproduced the error I was doing the following (which I think you were too):
ApplicationData::Current::GetPublisherCacheFolder("foo");
Try doing this:
ApplicationData::Current->GetPublisherCacheFolder("foo");

jjbenjohn, 2016-11-09:
@aronbardsley Sorry to jump in so late on this thread, but I'm having similar issues on IoT for a RPi3 project I'm currently working on.  Back in January, you said "This is is now working with a UI app that is deployed to the raspberry pi. Selecting the file type extensions worked as expected (for the most part, see below). I can now access removable storage on the Raspberry Pi."

For my project, I'm also running headless, have developed (and am enhancing) a small web browser for the Pi, am running a background task, and have a need to create various logs on one or more USB thumb drives.

Can you share a bit more detail on how you implemented your solution because I'm still have some difficulties combining a blank UI with the background tasking I'd like to use.

Thanks,
John

aronbardsley, 2016-11-16:
Hi John,

In the scenario I am working with, my app is the only one running on the device. I don't need to worry about hibernation or resume power states - which may be a factor in why you are attempting to use a background task? 

I created a new UI app and I run everything by using an async task:

`                Task ta = new Task(() =>
                {
                    BackgroundController server = new BackgroundController();
                    server.Run();
                });

                ta.Start();
`
This code was added to the OnLaunched method of the app class after the line that says "Window.Current.Activate()"

I use a task so I'm not blocking the main UI thread - otherwise the app will restart every few seconds. In my Run method, I've got my main request loop which listens on a socket as well as a bunch of other things (such as timer based activities). 

Perhaps you could use a similar approach by using tasks with your browser / UI app to perform the background tasks?

hubert-associates, 2016-11-27:
Hi, any word on a fix for this problem? I can access an FDTI  USB serial device from my IoT-Core/UWP/C#/Background-Task fine. Is there now a supported way to read a simple .txt from a USB-flash-drive in one of the USB ports? We need this, since there is no support for getting the unique hardware-ID from an iot-core/UWP ( RPi ). I can get the ID fine if I use linux on the same board. Anyway, as silly as it sounds, we are putting the device ID on a USB-stick and trying to read it from there... Any help for (any of these) problems welcome.

munyirik, 2016-11-28:
The workaround suggested in http://stackoverflow.com/questions/34754668/access-denied-error-when-accessing-usb-stick-files-in-windows-iot-core-app worked for me.
You need to create an empty headed app, add it as a reference to the headless app, then add 'Executable' and 'EntryPoint' of the headed app to the headless app manifest (see image below)
[image]

Also remember that you need to have the 'removablestorage' capability and file type association for txt files declared in the headless app manifest.

hubert-associates, 2016-11-28:
Thanks, I'll try this. Had some timing problems with headed, but will try again. Could you possibly share your manifest? since the syntax is not quite intuitive. :-)

munyirik, 2016-11-28:
Here it is
[Package.appxmanifest.txt](https://github.com/ms-iot/ntvsiot/files/617448/Package.appxmanifest.txt)

ummarbhutta, 2017-07-19:
Any update on this? I want to store sqlite database file in external storage, extension of file is .sqlite. I am using Windows IoT Core on RPi 3 with background application. I don't want to add a useless foreground application to my existing structure for this, any other proper solution? One I know is to use the well known media extensions like .jpg, .png (of course adding suitable capabilities as well). Any other fix for windows IoT?

paulmon, 2017-07-19:
The workaround is still required.  Your patience while we work to fix this pain point is appreciated

lprichar, 2018-01-11:
Any updates to this issue?  It's been two years since the original issue and the workaround is kind of a pain for deployment.