ms-iot/ntvsiot #73: Refactor deploy and bootstrapping code

alanch-ms, 2016-01-27:
"The deploy and bootstrapping code is getting hard to read with all the if conditions. Not sure how possible is this, but bonus points to you, if you can refactor all the deploy / boostrapping code into virtual member functions of (class) Target. Then the projectcfg class can call the virtual function, this->currentTarget->Deploy()."

nobodykid, 2016-07-07:
I got this error during the deployment process 

Error	 DEP6701 : Bootstrapping failed with unexpected error: 'Unable to load one or more of the requested types. Retrieve the LoaderExceptions property for more information.'.	NtvsUWPTest

munyirik, 2016-07-07:
Hi @nobodykid. Can you share these details:
1. Version of Visual Studio do you have (e.g. Visual Studio Community Update 2).
2. Version of Node.js Tools for UWP Apps (I'm assuming it's 1.5.3).
3. Type of device you are deploying to and the Windows build number.
Thanks.

nobodykid, 2016-07-07:
I'm currently using Visual Studio Community 2015 Update 3 with the latest Node.js Tools for UWP (1.5.3)

I tried to deploy this to my Raspberry Pi 2 B with Windows 10 IoT Core Insider Preview

munyirik, 2016-07-11:
I'll try and reproduce this today. Can you confirm that the build number for the insider build you have is 14376?

munyirik, 2016-07-11:
I just tried with 14376 and didn't see this problem. I saw online that a similar error has been seen ([link](https://forums.xamarin.com/discussion/70249/after-visual-studio-update-3-xamarin-4-1-1-3-cannot-deploy-to-windows-uwp-phone-or-8-1-phone-emula)) and it may be an issue with the SDK. Can you try installing the standalone Windows 10 SDK from [here](https://developer.microsoft.com/en-us/windows/downloads/windows-10-sdk) and see if you still have the problem?
Note:
After installation you should open NtvsUWPTest.njsproj in text editor (or unload the project and edit the file in VS). Then change the WindowsTargetPlatformVersion and the WindowsTargetPlatformMinVersion to the version of the latest SDK. This number is set automatically when a new project is created so you don't need to do this for new projects (only existing projects).

munyirik, 2016-07-11:
Also I'll create a different issue since this one tracks work for code refactoring.

IoTGirl, 2017-01-23:
Hi Munyiri,
Issue #100 has been closed. Can we get a status on this issue?
Sincerely,
IoTGirl

munyirik, 2017-01-23:
Closing this. For now, only critical bugs will be fixed.

closed 2017-01-23 by munyirik