ms-iot/ntvsiot #80: Error in deployment

rlyle78, 2016-02-02:
I'm getting this error when I deploy to Raspberry Pi my CylonJS project.

I know the file exist, I verified it. 

Severity	Code	Description	Project	File	Line	Suppression State
Error		Error : DEP2100 : Cannot copy the file "C:\CylonUwp1\CylonUwp1\node_modules\cylon-firmata\node_modules\firmata\node_modules\serialport\node_modules\node-pre-gyp\node_modules\minimatch\node_modules\brace-expansion\node_modules\balanced-match\example.js" to the target machine "192.168.43.149". 
The system cannot find the path specified. (Exception from HRESULT: 0x80070003)	CylonUwp1			
Help

Ron

munyirik, 2016-02-02:
Hi @rlyle78 do you have the latest version of the tools installed? i.e. v1.5.1?

munyirik, 2016-02-02:
That error is happening because the file path for example.js is [too long](https://github.com/Microsoft/nodejstools/issues/69) to be deployed to the device. To remedy this problem (if you have version 1.5.1 installed), right-click on the npm node in Solution Explorer and then click on 'Update npm Packages'. This will run "npm dedupe" which should then move node-pre-gyp module under CylonUwp1\node_modules to resolve this problem. Let me know if you have any issues doing that.

munyirik, 2016-02-04:
I'll close this now. @rlyle78 if you have any more questions please let me know. Thanks!

munyirik, 2016-02-09:
I was able to reproduce this issue and my previous instructions wouldn't solve the problem. If you only have J5 installed then you're fine. BUT if you have packages that are different versions of packages that node-pre-gyp depends on, then there's a problem. node-pre-gyp won't flatten out with the npm dedupe command. I found a workaround (which isn't great) but it worked for me. 
* I deleted the node_modules folder in the project
* Right click on the Johnny-Five node in the Solution Explorer window and click 'Install Missing Npm Package(s)'
* Right click on the npm node in Solution Explorer window and click on 'Update npm Packages'

I'll reopen this to investigate a solution.

munyirik, 2016-07-15:
Now that there's a way to remove the path length limit on Windows, this may not be a problem for much longer - https://news.slashdot.org/story/16/05/31/0012222/microsoft-removes-260-character-path-length-limit-in-windows-10-redstone

dick, 2016-08-01:
Any tips on avoiding the filename length error on NTVS 1.5.4? I upgrade to try the SerialPort fixes, followed your comment above about deleting node_modules, and I'm still getting this error:

Severity	Code	Description	Project	File	Line	Suppression State
Error		
DEP2100 : Cannot copy the file "C:\Users\dick\documents\visual studio 2015\Projects\petropi5\petropi5\node_modules\node-pre-gyp\node_modules\request\node_modules\har-validator\node_modules\is-my-json-valid\node_modules\generate-object-property\node_modules\is-property\is-property.js" to the target machine "192.168.2.106". 
The system cannot find the path specified. (Exception from HRESULT: 0x80070003)	petropi5

munyirik, 2016-08-01:
Hi @dick, Can you try running "npm dedupe" in C:\Users\dick\documents\visual studio 2015\Projects\petropi5\petropi5\node_modules?

When I do that on my machine it flattens the modules in the path you have. i.e. I see -
[my project]\node_modules\request
[my project]\node_modules\har-validator
[my project]\node_modules\is-my-json-valid
[my project]\node_modules\generate-object-property
[my project]\node_modules\is-property

dick, 2016-08-02:
@munyirik npm dedupe in the folder didn't help.

I tried running through your steps above again (starting with deleting the node_modules folder), with one slight change from the first time I did it. After selecting "Install missing npm packages," Visual Studio prompts me to run npm dedupe. This time I told Visual Studio to ignore the issue, then selected "Update npm packages." Npm dedupe had a lot of text output like ran correctly, and those modules do show up at the top level of my path, but the request module is still included under node-pre-gyp\node_modules with the deep nesting for other requirements.

munyirik, 2016-08-03:
If the request module is still in node-pre-gyp then is there another version that is in [my project]\node_modules\request? If there's a version of request at the root and node-pre-gyp depends on a different version of request then what you're seeing is normal. 
In this case you can delete the request directory in petropi5\node_modules\node-pre-gyp\node_modules. 
This will mean that node-pre-gyp will use the request package that's in the root of the node_modules folder. I've done this before and didn't have any problem with using a different version. Can you try that?
P.S. I'm out of the office at the moment but when I get back I'll look into getting instructions to remove the path limit and resolve this issue. Thanks for your patience.

LabbiqX, 2016-08-05:
I was getting the same error, but I solved the issue. 
The main problem is a bug of "npm dedupe". The dedupe command can not reduce all the duplicates.

To solve the problem follow these steps.
1. Run "npm dedupe" command in "PROJECT_DIR\node_modules".
2. Deploy the project and see the error. Find "PROJECT_DIR\node_modules\A_MODULE\node_modules\\....."
3. Run "npm dedupe" command in "PROJECT_DIR\node_modules\A_MODULE\node_modules"
4. Deploy the project without any error.

This is the absolute solution.

munyirik, 2016-09-06:
Update: While the latest release doesn't fix this problem we are actively working to resolve the long path limit on Windows IoT Core.

munyirik, 2016-09-13:
@rondagdag et al, 
You can enable long paths on IoT Core by running 

*reg add "HKLM\SYSTEM\CurrentControlSet\Control\FileSystem" /t REG_DWORD /d 1 /v LongPathsEnabled /f* 

Note that a reboot may be required after setting the registry key. This will work on the latest IoT Core build [14393](https://developer.microsoft.com/en-us/windows/iot/docs/releasenotesrtm). Setting this reg key should get rid of the *DEP2100 : Cannot copy the file ... to the target machine* error. Let me know if it doesn't.

closed 2016-09-26 by munyirik