ms-iot/ntvsiot #84: NTVS 2.1 Alpha nodejs universal app project templates not listed

michaelderemer, 2016-04-01:
My Env:
Latest version of iot 10586 on my raspie2 
Visual Studio 2015 version 14.0.25123.00 Update 2
Ran prereqs listed here:  https://github.com/ms-iot/ntvsiot
Installed NTVS 1.2 Alpha VS  2015.msi
Installed node-chakra-x86.msi

With NTVS 1.2 Alpha i no longer see any nodejs projects that are "universal app project"

Granted i have installed/Uninstalled about every combination of node tools. With 1.5.2 i could create the project, install npm uwp but they wouldn't deploy. I always got a very vague unable to start error message.

munyirik, 2016-04-01:
Hi @michaelderemer, could you check in Control Panel\Programs\Programs and Features if Node.js Tools for UWP Apps is installed?
I'm wondering if somehow the standalone 1.2 Alpha install removed it.

P.S. I'm testing v1.5.3 at the moment (which has 1.2 Alpha) and will be releasing it soon.

munyirik, 2016-04-01:
@michaelderemer I missed the step that you were running the prereq build and debug steps. Are you trying to build and debug this project? If so are you saying you're not seeing the templates in the experimental instance of Visual Studio?

munyirik, 2016-04-01:
I'd be happy to help:-) 
Could you verify if the error is happening on deploy (Build menu->Deploy Solution) or when it's launched (Debug menu -> Start debugging)? this will help isolate the issue.
Also is there an error code with the error message?

michaelderemer, 2016-04-01:
I got a c# universal app to deploy! 
I cannot get the nodejs webserver universal app to deploy though still. 
Deployed using Debug-> Start Debugging
Alert says Unable to start debugging. Operation not supported. Unknow error 0x8007057
Output windows says: DEP 3310: Unable to deploy an x86 package to an ARM device. Change the build config to ARM or Nuetral, or change the debug target to an emulator or attach an x86/64 device. DEP6953: Failed to launch remote debugger... command failed 0x80270254

The odd thing is at the top i have it set to DEBUG ARM Remote Machine. Also if i right click my project and go to properties its set to 
Configuration: Debug   Platform: Arm and then i have my remote machine configured to the name with node arg --debug

Not really sure why its compiling x86 when it knows my target is ARM. What am i missing?

munyirik, 2016-04-01:
Strange. Looks like you ran into https://github.com/ms-iot/ntvsiot/issues/23 but it was fixed a while back. Can you try 'Rebuild' first and then try and deploy?

michaelderemer, 2016-04-01:
I was able to get a brand new nodejs webserver universal app project to deploy once. However when I included the uwp npm package and pasted in the nodeblinky code. The deploy failed several times.

I repasted the original nodejs webserver code and saved. 

 

Try 1) Build btn> rebuild > deploy > -No errors in output or alerts just deploy failed in lower left.

Try 2) Build btn > Clean Build > build > deploy – no errors at all just deploy failed

Try 3) Build btn > rebuild > Debug btn > Start debugging

Resulted in: Unable to start debugging operation not supported. Unknown error 80070057

DEP6953: Failed to launch remote debugger with the following error: Failed to launch remote debugger with the following error: Command failed 80270254

 

I also tried:

right clicking the solution and rebuilding solution > deploy

Clean solution > build solution > deploy

Rebuild solution > F5

 

Then I tried starting a new project and this time I got the same errors as above. 

Anyways this has consumed my last couple days and I am getting behind on real work. I may have to wait till the tools stabilize.

munyirik, 2016-04-01:
Sorry about issues you're running into. And thanks for detailed info. Couple of things.

1. I'll update the "Build and debug" section in the README. I can see how the instruction there can be taken to mean building and debugging Node.js apps (the instructions are only mean for contributors to build and debug the code in this repository).

2.  The uwp npm package is actually included automatically when you are using any of the Universal templates. The version included is actually built differently from the one you get when you do "npm install uwp" which works with the console version of Node.js (Chakra). To use the Windows.Devices.Gpio namepace you only need to do: "var uwp = require('uwp'); uwp.projectNamespace('Windows');" - This should also probably be documented better.

3. I was trying to reproduce your problem and also got some deploy errors with 1.2 Alpha. As part of the 'bundle' installer, Node.js (Chakra) console app (i.e. node.exe) is also installed. npm which is used by Visual Studio comes as part of this installation. Currently there is a bug where it cannot work with typings or tsd. E.g. if you try "typings install winrt-uwp --ambient" this will fail.
1.2. Alpha makes these calls automatically and when they fail I think it's affecting the build but not exactly sure how. I was able to work around the problem by installing [node v8](https://nodejs.org/en/). As one last favor you can see if this works for you:-). 
  * Install node v8 
  * update path environment variable so that C:\Program Files\nodejs comes before C:\Program Files\NodejsUwp\Console
  * Restart Visual Studio.
  * Try creating a new Node.js Web Server (Universal) and see if you can build and deploy.

4. I the workaround above fails, please send me a mail at munyirik@microsoft.com and I can set up a call to work through the issue.

5. We'll have another version of Node.js for UWP Apps coming soon which includes 1.2 Alpha which should fix this issue.

Thanks!

munyirik, 2016-04-18:
closing since the 1.5.3 release resolved this issue.

closed 2016-04-18 by munyirik