ms-iot/ntvsiot #91: Deployment to Windows Phone 10

AmadeusW, 2016-04-15:
Hi, I'm trying to deploy  [‘Hello, World’ Node.js (Windows Universal) Sample](https://ms-iot.github.io/content/en-US/win10/samples/NodejsWU.htm) sample app to Windows Phone OS 10.0.14295

Whether I provide my phone's IP or leave `Remote Machine` blank, I get this error when I deploy. I couldn't find anything about it on line.
```
DEP0001 : Unexpected Error: -2147483638	
```
[image]

Is Windows Phone supported? How do I specify phone as the target device for universal windows node.js projects? Is there a tutorial I can follow to get it working? 

I'm really stoked to get this working on the phone!

AmadeusW, 2016-04-15:
Update: I enabled [Embedded Mode](https://ms-iot.github.io/content/en-US/win10/EmbeddedMode.htm) on both PC and phone. Now when I'm deploying to Remote Machine using the phone's IP it asks me for PIN.

I tried both my device PIN as well as a code under **Device Discovery -> Pair**, it didn't work.
[image]

In phone settings -> For Developers, I set **Device Discovery** to **On** and **Device Portal** to **On**. I tried with **Authentication** either **On** or **Off**, this didn't help.

Then I added a flag into the registry `reg add HKLM\SYSTEM\CurrentControlSet\Control\SirepClient /v DisableProtocol3 /t REG_DWORD /d 1` [as shown here](https://stackoverflow.com/questions/33473200/how-to-develop-windows-10-uwp-on-real-device-mobile/33667162#33667162)

Now, deploying to **Remote Machine** gives this error:
```
DEP6956 : Expecting Ssh transport protocol, got 'Sirep2'.	NodejsWebServer1			
```

Deplying to **Device** gives this error, despite developer mode being enabled on both phone and PC.
```
DEP0100 : Please ensure that target device has developer mode enabled. Could not obtain a developer license due to error 800704C7.
```

munyirik, 2016-04-15:
Hi @AmadeusW,
Yes, deploying to PC or Phone is supported. For the deployment target for Phone you need to use Device (not Remote Machine). The PIN entry is something you shouldn't need to do...and it doesn't work:-). To get it working you need to do the following is done:

1. Enable Embedded Mode (looks like you already did this).
2. Settings->Update & Security->For Developers->Developer Mode set.
3. Settings->Update & Security->For Developers->Device Discovery On.

Can you check if this is the case?
Thanks.

AmadeusW, 2016-04-15:
The settings are set like you specified. I'm now deploying to the device.
I reverted the registry flag and updated Universal Tools to v1.3.1.
This gets me one step further:

[image]

Debug output is empty and a breakpoint wasn't set.
Could it be that the app quits since it has no UI?

munyirik, 2016-04-15:
Can you try with the latest tools ([1.5.3](https://github.com/ms-iot/ntvsiot/releases/tag/v1.5.3))? v1.3.1 didn't support deployment to phone.
The app will quit if you don't have code that's waiting (e.g. if you're just doing a console.log vs. having a server waiting for requests).
You shouldn't see that error message ('Unable to activate...') if everything is working correctly.

AmadeusW, 2016-04-15:
Oh, I meant this v1.3.1 (and VS2015 update 2)
[image]
It still didn't work with the same error.

I've tried on another pc that has VS2015 update 2, Universal tools v 1.2 and Node.js Tools for UWP Apps 1.5.3
It also didn't work, with the same error. 

I guess my phone is the culprit. Here are the settings:
[image]
[image]

munyirik, 2016-04-16:
Could you tell me what the OS build number is for the phone? You can get it in Settings -> System -> About -> More Info.

AmadeusW, 2016-04-16:
BLU WIN HD W510u
Version: 1511
OS build: 10.0.14295.1000

munyirik, 2016-04-16:
Ah! looks like you ran into a issue that's been fixed on newer builds. I'll find out if it's possible to get you an update...

AmadeusW, 2016-04-16:
Nice! I'm on the fast ring of insider updates.

munyirik, 2016-04-16:
Is 14295 the newest insider build you have access to?

AmadeusW, 2016-04-16:
It works like a charm on 14322 :) Thank you for help!

Can you tell me more about application lifecycle on the phone? How do I turn the app off? How do I turn it on? Is there a sample that provides a basic ui? I really just need a way to turn the app on or off, and perhaps display a status message on the phone's screen.

closed 2016-04-16 by AmadeusW

AmadeusW, 2016-04-16:
I'm looking for a sample with a UI because I need to get a permission to use capabilities like geolocation, and of course I would like to control when the app is on and off. 

I tried to incorporate elements of [Windows-universal-samples - Geolocation](https://github.com/Microsoft/Windows-universal-samples/tree/master/Samples/Geolocation) into this server but couldn't get geolocation to work (maybe it needed permissions?), and conversely I couldn't inject the node.js server into that sample because `Manifest references file 'nodeuwp.dll' which is not part of the payload.` and I couldn't figure out how to include it.

munyirik, 2016-04-17:
The application is a [background/startup application](http://ms-iot.github.io/content/en-US/win10/BackgroundApplications.htm) (see more info in the link) so it's not designed to have a UI. The way an application would be set up is with the [iotstartup tool](https://ms-iot.github.io/content/en-US/win10/tools/CommandLineUtils.htm) and would be stopped with kill.exe.
If you want to display something on the screen you would need to do it from a [regular XAML app](https://msdn.microsoft.com/en-us/windows/uwp/get-started/create-a-hello-world-app-js-universal). Would that work for what you are trying to build?
To enable geolocation you just need to add the capability in your Package.appxmanifest file. So you would have something like this:
```xml
...
  <Capabilities>
    <!-- DeviceCapability elements must follow Capability elements (if present) -->
    <DeviceCapability Name="location"/>
  </Capabilities>
...
```

AmadeusW, 2016-04-20:
Thanks. I think I get it: the application is launched by the OS at startup and does its work without any user interaction. 

How do you communicate with a background task? The IPC is quite limited in Universal Apps, but I saw that [App Service](https://msdn.microsoft.com/en-us/windows/uwp/launch-resume/how-to-create-and-consume-an-app-service) is also a background app that can process a `Request` and return a response with [AppServiceConnection.RequestReceived](https://msdn.microsoft.com/en-us/library/windows/apps/xaml/windows.applicationmodel.appservice.appserviceconnection.requestreceived.aspx). 

Is there a way to have a node.js server running within a Universal App that is not a background task?

Ultimately, I want to have a web server and I want to turn the location on and off to conserve the battery.

AmadeusW, 2016-04-20:
Okay, I think I get it. It would be great to have node.js running in a headed app that user can install from the store, run, close and uninstall. Thanks for help!

munyirik, 2016-04-20:
Thanks for the feedback @AmadeusW. That's something we are considering.