ms-iot/ntvsiot #92: Access is denied when accessing geolocation data

AmadeusW, 2016-04-20:
Headless mode (Background task) doesn't ask the user for permission to use the location, and hence we get an error on `getGeopositionAsync()`:
```
WinRTError: Access is denied.

Your App does not have permission to access location data. Make sure you have defined ID_CAP_LOCATION in the application manifest and that on your phone, you have turned on location by checking Settings > Location.
```
Adding `<DeviceCapability Name="location"/>` is not enough. User needs to go to Settings and allow the app to access location:
[image]

Ideally, the OS would ask the user for permission to access location the first time the background task starts and attempts to access location.

munyirik, 2016-04-20:
This is by design for side-loaded apps like IoT background apps. If an app is installed through the store then I think that's when the OS would ask for permission. Let's keep this open until I've confirmed.

munyirik, 2016-07-05:
Sorry for the delay. This looks like it's by design at least for now. With a UI (headed) app the call to Windows.Devices.Geolocation.Geolocator.requestAccessAsync the OS is supposed bring up a prompt. For a background app (headless) it's not designed to.

closed 2016-07-05 by munyirik