ms-iot/ntvsiot #93: serialport list ports returning undefined manufacturer, etc.

dick, 2016-04-26:
I managed to get serialport working with a Raspberry Pi 2 and Windows IOT with some teeth-gnashing (will open another issue for serialport.node being searched for in wrong folder).

Now that the module is loading, serialport.list() is not responding the way I expect it to.

Code:
`var serialPort = require('serialport');

    var serialData = {};

    serialPort.list(function (err, ports) {
        if(err) { 
            console.log(err); 
        }
        else {
            console.log("Ports " + JSON.stringify(ports));
            ports.forEach(function (port) {
                console.log("Port!\n",port,"\n\n");
            });
        }
    });`

Console output: 
`Ports [{"comName":"\\\\?\\FTDIBUS#VID_0403+PID_6001+FTGDNS37A#0000#{86e0d1e0-8089-11d0-9ce4-08003e301f73}"},{"comName":"UART0"}]
Port!
 { comName: '\\\\?\\FTDIBUS#VID_0403+PID_6001+FTGDNS37A#0000#{86e0d1e0-8089-11d0-9ce4-08003e301f73}',
  manufacturer: undefined,
  serialNumber: undefined,
  pnpId: undefined,
  locationId: undefined,
  vendorId: undefined,
  productId: undefined } 

Port!
 { comName: 'UART0',
  manufacturer: undefined,
  serialNumber: undefined,
  pnpId: undefined,
  locationId: undefined,
  vendorId: undefined,
  productId: undefined }`

munyirik, 2016-04-26:
Hi @dick  and sorry about the teeth-gnashing. There's still work to do to get npm aware of the UWP platform. For now you should be able to get the right serialport by 'right click npm node in Solution Explorer -> Update npm packages...' That should be putting serialport.node in the right folder if you're using > v2.1.0. Please open an issue if it's not.
The information I provide is based on the [SerialDevice properties I can access](https://msdn.microsoft.com/en-us/library/windows.devices.serialcommunication.serialdevice.aspx). In my initial commit I only included PortName (which is what you see) but I should also include UsbProductId and UsbVendorId which are also available.
I'll have to do some research on how to get the other properties.

dick, 2016-05-15:
Hi @munyirik thanks for the info.

Is comName the proper parameter to pass to serialport when creating a new port? serialport.list is properly finding 'UART0' for me, but I get a mysterious "The program '[2496] backgroundTaskHost.exe' has exited with code 3 (0x3) error" when serialport tries to open a port with 'UART0' as the input.

I have had success opening a port by assigning a port to "COM0" using the registry instructions described here: https://developer.microsoft.com/en-us/windows/iot/win10/samples/nodejscylon but it seems overly complicated to have to make registry configurations for every port...

munyirik, 2016-05-15:
'UART0' should work so I think that's a bug. I'll try it out when I get in the office tomorrow.

dick, 2016-05-15:
Thanks! One more piece of info -- I'm having similar experience with device strings as well. Would you expect "FTDIBUS\VID_0403+PID_6001+AH02X2K9A\0000" to be a valid input?

munyirik, 2016-05-15:
Yeah, that's valid but after looking at the code that parses the name (see [this](https://github.com/munyirik/node-serialport/blob/uwp/src/serialport_uwp.cpp#L26)), it will fail. The code expected the device string to start with "USB". I can fix that.
In the meantime, can you communicate with the device if you manually assign a port name to it?

dick, 2016-05-16:
Yes, that one also seemed to open when I manually assigned a port name.

munyirik, 2016-05-16:
Ok, I know what's going on with UART0. The way the names are parsed is a little broken. You should be able to just copy the name you get with list() and paste it in open() which isn't happening. I'll fix this for the next release.

karlssja, 2016-06-14:
The following code works for me (I am using a FTDI USB to Serial converter)
```
// Serialport
var comName = 'USB\VID_0403&PID_6001\A10121CF';
//var comName = 'UART0';

var serialport = require('serialport');
serialport.list(function (err, ports) {
    for (var ii = 0; ii < ports.length; ii++) {
        comPorts[ii] = ports[ii].comName;
        if (ports[ii].comName.startsWith('FTDIBUS#VID_0403+PID_6001+A10121CF',4)) {
            comPort.open();
        }
    }
});
```
Please note that the comName used to open the port and reported is different. The Device manager says ID : USB\VID_0403&PID_6001\A10121CF

Changing it to UART0 gives a code 3 (0x3) error. It would have been nice to be able to use the internal UART to.

munyirik, 2016-06-28:
FYI currently working on this. UART will be fixed too.

munyirik, 2016-07-15:
The serialport issues should be fixed in latest release. Let me know if you run into any other issues. 
Things to note:

* You'll still need to run the 'Update npm packages...' command to get the latest UWP serialport with the fix. In future I'll be working on issue #102 so that the patch isn't required anymore.
* manufacturer is only going to be [listed if it's available](https://github.com/munyirik/node-serialport/blob/uwp/src/serialport_uwp.cpp#L490).

closed 2016-07-15 by munyirik

karlssja, 2016-07-15:
Latest fix returns errors

Using MS VS 2015 Update 3
npm install serialport 4.01 

Download complete.
Extracting serialport UWP patch.
Extraction complete.
**Error. C:\Users\...\serialportpatch\uwp\serialport.js does not exist**
Copying C:\Users\...\serialportpatch\uwp\ARM\serialport.node to C:\Users\...\node_modules\serialport\build\Release\serialport.node
serialport UWP patch applied!

var SerialPort = require("serialport");
**The program '[3608] backgroundTaskHost.exe' has exited with code 1 (0x1).**

munyirik, 2016-07-15:
Hi @karlssja. Please install https://github.com/ms-iot/ntvsiot/releases/tag/v1.5.4. It should fix that error.

munyirik, 2016-07-15:
Also backgroundTaskHost exit code 1 can be ignored for now. node UWP currently doesn't pass it's error code to backgroundTaskHost so it's not useful for debugging node errors. It should be 0 if nothing went wrong though. It's a known issue I haven't yet looked into.

munyirik, 2016-07-15:
One more thing. For you device, use the string "\\\\?\\FTDIBUS#VID_0403+PID_6001+A10121CFA#0000#{86e0d1e0-8089-11d0-9ce4-08003e301f73}"
i.e. double the backslashes.

karlssja, 2016-07-15:
Installing v1.5.4 fixed my errors!
Now the internal serialport UART0 on my Pi2  works!
also the FTDI usb adapter works after changing the comName as you suggested!

Brilliant job!!! Thank You!

munyirik, 2016-07-15:
Glad to hear it. Also I've opened an issue about the exit code and I'll fix that soon - https://github.com/ms-iot/ntvsiot/issues/104