ms-iot/ntvsiot #94: the procedure entry point JsInstanceOf could not be located in the dynamic link library

stanleyoho, 2016-05-06:
hello i can't excute Node.js (chakra) command prompt.

when i open it , will show this error message "the procedure entry point JsInstanceOf could not be located in the dynamic link library C:\Program File\NodejsUwp\Console\node.exe"

anyone know this ?

munyirik, 2016-05-06:
Hi @stanleyoho,
What version of Node.js (chakra) do you have? You can check the 'Version' column in 'Control Panel\Programs\Programs and Features'
Also can you tell me what version of Windows you have? (you can copy/paste the output from the command `systeminfo | findstr /B /C:"OS Name" /C:"OS Version"`).

munyirik, 2016-06-29:
@stanleyoho I'm closing this for now. Please re-open if this is still an problem for you.

closed 2016-06-29 by munyirik