ms-iot/ntvsiot #97: typings : typings ERR! message Unable to find "winrt-uwp" ("npm") in the registry.

modembug, 2016-06-10:
PM> typings install winrt-uwp --global
typings : typings ERR! message Unable to find "winrt-uwp" ("npm") in the registry. Did you want to try searching another source? Also, if you want contribute these typings, please help us: https://github.com/typings/registry
At line:1 char:1
+ typings install winrt-uwp --global
+ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : NotSpecified: (typings ERR! me...ypings/registry:String) [], RemoteException
    + FullyQualifiedErrorId : NativeCommandError
 
typings ERR! caused by https://api.typings.org/entries/npm/winrt-uwp/versions/latest responded with 404, expected it to equal 200

**This can be fixed by running typings install dt~winrt-uwp --global**

Might want to update the installation documentation.

munyirik, 2016-06-10:
Hi @modembug, the command in the documentation (from notes in the last release) is "typings install winrt-uwp --ambient" - Is there another place where you saw the "--global" command?

modembug, 2016-06-10:
From what I read ambient has been replaced by global. If you do not specify the source with dt~ it will throw the error as it defaults to npm apparently.

Sent from Nine<http://www.9folders.com/>

munyirik, 2016-06-11:
Ah okay. What version do you have?

modembug, 2016-06-11:
Of visual studio? 2015

Sent from Nine<http://www.9folders.com/>

munyirik, 2016-06-11:
I meant typings but nevermind. I saw what you stated on [their page](https://www.npmjs.com/package/typings) that --ambient is now --global. Thanks for the tip @modembug !!

munyirik, 2016-06-12:
I've updated the instructions on https://github.com/ms-iot/ntvsiot/releases/tag/v1.5.3

closed 2016-06-12 by munyirik