ms-iot/ntvsiot #100: Error DEP6701 : Bootstrapping failed with unexpected error: 'Unable to load one or more of the requested types. Retrieve the LoaderExceptions property for more information.

munyirik, 2016-07-11:
Issue to track bug that @nobodykid is seeing where the error in the title is seen when attempting to deploy.
@nobodykid's environment:
Visual Studio Community 2015 Update 3
Node.js Tools for UWP (1.5.3)
Raspberry Pi 2 B with Windows 10 IoT Core Insider Preview

My last note:
I just tried with 14376 and didn't see this problem. I saw online that a similar error has been seen (link) and it may be an issue with the SDK. Can you try installing the standalone Windows 10 SDK from here and see if you still have the problem?
Note:
After installation you should open NtvsUWPTest.njsproj in text editor (or unload the project and edit the file in VS). Then change the WindowsTargetPlatformVersion and the WindowsTargetPlatformMinVersion to the version of the latest SDK. This number is set automatically when a new project is created so you don't need to do this for new projects (only existing projects).

munyirik, 2016-07-15:
Closing for now. If reinstalling the SDK didn't work please let me know.

closed 2016-07-15 by munyirik

nobodykid, 2016-07-22:
Solved the problem, and it's because my (somehow broken) VS. Besides these, I had several problem, such as unable to open appx.manifest on any UWP template.

I did repair on VS, reinstalling everything (NTVS and NTVS-UWP), and it worked.

Sorry for the late update

munyirik, 2016-07-22:
Glad to hear it. Thanks for the update.