ms-iot/ntvsiot #101: Error when trying to unstall uwp package

StrahilKazlachev, 2016-07-12:
I get the following error when I `npm install uwp`
`C:\Program Files\NodejsUwp\Console\node_modules\npm\node_modules\node-gyp\src\win_delay_load_hook.c(34): error C2373: '
__pfnDliNotifyHook2': redefinition; different type modifiers [e:\dev\samples\node-uwp\node_modules\uwp\build\uwp.vcxpro
j]`
* ntvsiot 1.5.3
* VS Community 2015 Update3
* Tried it both on Win10 with latest updates and the latest Win10 Insider Preview

I did find this issue nodejs/node-gyp#972, but I can't change the npm in a "working manner". Any ideas?

munyirik, 2016-07-12:
Hi @StrahilKazlachev, so running "npm -g install npm@next " didn't work for you?

StrahilKazlachev, 2016-07-12:
I did try, but the npm version stayed 3.8.3. I can't manage to change the npm that node-chakra is using.

munyirik, 2016-07-12:
Ok, I'll look into this...

munyirik, 2016-07-12:
BTW, I'm assuming you're going to use the uwp package with the console version of node. If you're using node UWP (i.e. creating the app from Visual Studio) then it (uwp.node) will be packaged automatically in your app.

StrahilKazlachev, 2016-07-12:
Will try it that way. I wanted do a simple console demo/sample.

munyirik, 2016-07-12:
FYI. This will be fixed in the release for this week. It was fixed in the node branch with this commit - https://github.com/nodejs/node-gyp/commit/f31482e2260e3beb90d959c62135e5e95c06ed4d
You can make the change yourself if you need it asap. The changes in the commit correspond to the following files:

- [program files]\NodejsUwp\Console\node_modules\npm\node_modules\node-gyp\src\win_delay_load_hook.c
- [program files]\NodejsUwp\Console\node_modules\npm\node_modules\node-gyp\addon.gypi

StrahilKazlachev, 2016-07-13:
Thank you. For now I'm good using VS as you suggested.

munyirik, 2016-07-15:
Issue is fixed in the latest release

closed 2016-07-15 by munyirik

StrahilKazlachev, 2016-07-15:
Works nicely.