ms-iot/ntvsiot #107: error deploying basic johnny-five project on x86

paulmon, 2016-08-09:
I am seeing this error (below)
I have a zip file with the project that has issues, but it's essentially the output from the new project wizard

------ Deploy started: Project: j5b, Configuration: Release x86 ------
Creating a new clean layout...
Copying files: Total 26 mb to layout...
DEP8000 : Unexpected deployment failure : AggregateException : One or more errors occurred.
DEP0600 : The following unexpected error occurred during deployment: 
Failed to deploy through new deployment pipeline.

munyirik, 2016-09-02:
I'm running into this issue. My repro steps:
1. Create new J5 project.
2. Do 'Update npm packages' to run npm dedupe and patch serialport.
3. F5

When I do this I got a path too long error (I think this is what you hit too) - there are two copies of serialport, one in the node_modules root and one in the firmata package. When I fix the issue (remove the one in firmata), that's when I get the AggregateException error. I think the error is a side effect of something getting messed up with the long path.

munyirik, 2016-09-02:
So I rebooted my RPi and was able to deploy the app. Seems like the first long path error is what messes things up. But even though I can deploy now, J5 is not working for me (exits with error 0xc0000409). Also on about the 5th deploy my RPi rebooted. Seems like my device is messed up. @paulmon I think this is similar to what happened to you right?

paulmon, 2016-09-02:
The problems I was seeing were always fixed by rebooting the RPi.  When the path was too long the vs code extension would fail to copy them, and setting the reg key didn't fix deploying a file since server seemed to have a check for MAX_PATH

munyirik, 2016-09-07:
Some more info...
* This will repro with the Cylon template as well (so basically any project that causes the deployment error in https://github.com/ms-iot/ntvsiot/issues/80)
* The only workaround to work for me after getting the AggregateException is:
  * Restart Visual Studio
  * Reboot the target device
  * Open the project again and F5

munyirik, 2016-09-13:
Closing this. I'll provide long path updates with https://github.com/ms-iot/ntvsiot/issues/80

closed 2016-09-13 by munyirik