ms-iot/ntvsiot #119: Support for sqlite3 npm package

psled, 2016-11-03:
I'm installing native sqlite3 npm package with following command:

`npm install sqlite3 --save --target_arch=x64 --node_uwp_dll`

When I try to use that package I'm getting a runtime error (I guess because of I installed a standard Win32 version instead of UWP one):

```
A dynamic link library (DLL) initialization routine failed.
node_modules\sqlite3\lib\binding\node-v48-win32-x64\node_sqlite3.node
```

I'm also trying to install above package with additional `--build-from-source` option but it fails as well, this time during the build process:

```
...
Command line error D8048: cannot compile C file 'C:\Users\...\Documents\Visual Studio 2015\Projects\NodejsWebServer1\NodejsWebServer1\node_modules\sqlite3\build\Release\obj\global_intermediate\sqlite-autoconf-3150000\sqlite3.c' with the /ZW option [C:\Users\...\Documents\Visual Studio 2015\Projects\NodejsWebServer1\NodejsWebServer1\node_modules\sqlite3\build\deps\sqlite3.vcxproj]
...
```

Microsoft is officially supporting SQLite for UWP apps with a special build of this library (available as Visual Studio extension). According to this [page](https://msdn.microsoft.com/en-us/windows/uwp/data-access/sqlite-databases) there are at least some special WINRT preprocessor macros being used during the compilation time.

Are there any plans to support sqlite3 package for Node.js UWP (I guess again in similar way as serialport is being supported - repository fork)?

munyirik, 2016-11-03:
We didn't have plans to support it but I'll look into it in the next couple of days. I'll keep you posted. 

What kind of app are you building if you don't mind me asking?

psled, 2016-11-03:
Thanks for a quick reply.

My scenario might be quite specific and not really relevant to others. I'm only considering it as one of possible solutions (still not sure if it is a correct one).

I'm going to integrate a medium-sized local area network backend already written in Node.js (Express, REST, WebSocket, Database, SerialPort etc.) with native, also already written, UWP C++/CX GUI app.

At the moment, due to UWP loopback restriction policy, I'm required to have in local network at least one server machine (with Node.js backend running on it) and at least one client machine which connects to that server (these have to be separate machines). This is not an issue for bigger networks but sometimes it might be a good idea for that particular app to not have a local network at all and run everything on a single machine. And yes, I know there are some ways get around of this issue in LOB applications but that's not what I'm looking for (the GUI / combined app has to be available in a Windows Store).

Having a Node.js code running side by side with UWP GUI app in a same process could also benefit of multiplication of server's code into each of client (now also possibly a server) machines. Another benefit is a single package deployment.

This is more generic UWP issue rather than IoT, although utilizing a low cost ARM hardware with Windows IoT Core would be an additional, really welcome, bonus.

In the end I could rewrite backend completely into native code and abandon Node.js code but before I decide to do that it's good to know other viable options.

munyirik, 2016-11-10:
Hi @psled,
I had a chance to look into this and here's what I found:

1. If you want to get around the UWP loopback policy, you can. Use the following steps:
  * Run `reg add hklm\system\currentcontrolset\services\mpssvc\parameters /v IoTInboundLoopbackPolicy /t REG_DWORD /d 1` and reboot the machine.
  * Ensure that the app is loopback exempt. To do this run `CheckNetIsolation LoopbackExempt -a -n=<Your App ID>`. To show a list of loopback exempted UWP apps run `CheckNetIsolation LoopbackExempt -s`

2. I was able to build a version of sqlite3 that I could use in a UWP app, BUT there are issues that would need to be investigated/resolved. In case you want to try it out asap, here are the steps I used to build it:
  * Clone https://github.com/mapbox/node-sqlite3
  * cd to node-sqlite3 and run `node "C:\Program Files\NodejsUwp\Console\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js" configure --node_uwp_dll=true --target_arch=<arch> --node_engine=chakra --module_name=node_sqlite3 --module_path=.`
 * Open node-sqlite3\build\binding.sln in Visual Studio.
 * Right-click on sqlite3.c (in the sqlite3 project) and click on Properties. In Configuration Properties->C/C++->General, set Consume Windows Runtime Extension to `No` (this gets rid of the D8048 error you saw)
 * Copy the text in [sqlite.c_patch.txt](https://github.com/ms-iot/ntvsiot/files/582269/sqlite.c_patch.txt) to sqlite3.c. This step just adds declarations for functions not supported in UWP. If there's something you use that makes sqlite3 call these functions then the app will crash.
 * Build.
 * Then you can then replace the .node file in your app and try it out.

I was able to run the tests as well but I had to manually move some files to the app's local storage folder (looks like a bug in some of the node fs functions for UWP). 
Anyway, based on the results - [sqlite3_uwp_test_results.txt](https://github.com/ms-iot/ntvsiot/files/582244/sqlite3_uwp_test_results.txt), it looks like most things work. But even if the test issues are resolved I don't know the extent to which the functions that don't work in UWP will affect your usage.

psled, 2016-11-10:
Thank you very much for a help, @munyirik .

I'm not in rush with that integration process and I guess I will stick with a present solution for a moment.

However, it's good to know that it's technically possible to use sqlite3 inside UWP and I will try it in a meantime (based on you instructions).

munyirik, 2016-11-10:
@psled I've update the steps to build. Simply update the gyp file for sqlite3 and you can build and use it in your UWP app.

psled, 2016-11-11:
Just confirming that based on your instructions I was able to successfully run simple Node.js test app using sqlite3 inside UWP.

jensrad, 2017-02-27:
Hi munyirik,
how described in another Forum ;-) I've got the same Problem. I tried to follow your recommend steps. Unfortunately this was not working. So this is what I did:

- Clone [https://github.com/mapbox/node-sqlite3](https://github.com/mapbox/node-sqlite3) => OK
- Replace node-sqlite3\deps\sqlite3.gyp with sqlite3.gyp.txt (remove the .txt extension). => OK
- In cmd window, cd to node-sqlite3 and run `node "C:\Program Files\NodejsUwp\Console\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js" rebuild --node_uwp_dll=true --target_arch=<arch> --node_engine=chakra --module_name=node_sqlite3 --module_path=.`
=> Result of the last step is the message: `The system can not find the specified file`
=> then I replace the tag `<arch>` to `arm` there Comes up a lot of more activities

[image]

What did I do wrong? Thank you very much for help!!!

munyirik, 2017-03-06:
Sorry for the delay (I was out of the office). I'll respond today...

munyirik, 2017-03-06:
It looks like you didn't "cd to node-sqlite3". You need to run the command in the node-sqlite3 clone folder. The error in your log is that binding.gyp wasn't found which will normally be in that folder.

jensrad, 2017-03-07:
Hi, thanks for comming back. I hope you have had a nice holiday :-).
Yes you are right, I missed the tip to work inside the folder. Unfortunately there are other things going wrong. According to the picture the declaration of `node_engine_libs` is missing. Testwise I put the declaration to the file `binding.gyp`. After that then comes up the error that declaration of `node_engine_include_dir` is missing, too. When I put both variables to `binding.gyp` compiling goes further with new errors (for example: `cannot find module 'nan'`). Maybe these errors would disappear when I would use the right variable declarations (I don't know the correct declaration of these variables). 

Here the picture of error message before manipulating `binding.gyp`.

[image]

munyirik, 2017-03-07:
There was a missing step I my list. You also need to run "npm install nan" since you'll need nan to build   the node-sqlite3 module.

Another thing to confirm. If you run "where node" make sure that it's the node-chakra versionnnnn (C:\Program Files\NodejsUwp\Console\node.exe) that is first on the list.

Lastly, I've tested this with some newer builds on Windows and it will not build at all. The problem is actually related to the issue you saw with Express (Reflect.apply). But since I don't see that error in your log it doesn't apply to you. I'm hoping to resume the merge work (https://github.com/ms-iot/node/pull/14) I started and get a release out soon but I have no ETA at this time.

jensrad, 2017-03-07:
Thank you for Information. So with this last Information I could build. As summery what I've done:

- git clone [https://github.com/mapbox/node-sqlite3](https://github.com/mapbox/node-sqlite3)
- Replace node-sqlite3\deps\sqlite3.gyp with sqlite3.gyp.txt (remove the .txt extension)
- open cmd-Window 
   => check the order of node.js installations (C:\Program Files\NodejsUwp\Console\node.exe must come first) by the command: `where node`
   => in my case the order was wrong; because of this change order of Node.js entries in the Windows Environment variable (after changing open a new cmd-Window)
- within cmd-Window:
   => check Python Version by the command: `python`
   => in my case Python 2.7 was installed but there was no entry in the Windows Environment variable; if so put Installation path of Python to the Windows Environment variable (for example C:\python27)
- within cmd-Windows: 
   => goto the cloned node-sqlite3 folder
   => when run the command: `npm install nan`
- Finally in the cmd-Window run the command: `node "C:\Program Files\NodejsUwp\Console\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js" rebuild --node_uwp_dll=true --target_arch=arm --node_engine=chakra --module_name=node_sqlite3 --module_path=.`

The very last row of all activities during building is 
[image]

So tomorrow I'll try to use the build result within an Node.js Project for Windows IoT. Then I will give Feedback.

For the  (ms-iot/node#14) Task I will patiently wait for a solution...