ms-iot/ntvsiot #120: Node UWP App deployment blocked by group policy on Windows 10 Desktop

nVentis, 2016-11-29:
(The same question can be found [onStackOverflow](http://stackoverflow.com/questions/40795000/node-js-for-uwp-blocked-by-group-policy/40795026))

After having developed a small ("vanilla") Node JS application, I tried to deploy it to Windows 10 IoT. This worked successfully, but I'm experiencing an unknown error that leads to the app (and the remote debugger) instantaneously being closed / stopped.

However, the real problem ist, that whenever I try to deploy the same Node UWP app on my local Windows 10 Desktop device (in order to check where the error could come from), I'm getting a critical error that prevents me from deployment:

[image]

An English translation could be something like

> The Windows Store-App [...] can not be activated. The activation request has failed with error "The application was blocked by a group policy. Contact the system administrator for further information".

I'm not sure what the reason could be, but I'm wondering why the Windows Settings show `Searching developer package` whenever I check my `Developer settings` in the `Updates and Security` menu.

nVentis, 2016-11-29:
Please, do also note the following error I got from the event log

`Failed with 0x490 modifying AppModel Runtime status for package [...] for user [...] (current status = 0x0, desired status = 0x20).`

munyirik, 2016-11-29:
Hi, so if you're using to Windows 10 IoT Core you don't need to do anything extra to deploy [background apps](https://developer.microsoft.com/en-us/windows/iot/docs/backgroundapplications) (Node.js UWP runs as a background app). 
However, if you're targeting Windows 10 desktop or mobile, you need to enable Embedded mode (the error you see is because of this). Follow the instructions in the link [here](https://developer.microsoft.com/en-us/windows/iot/docs/embeddedmode) to enable Embedded mode.

munyirik, 2016-11-29:
I just saw that in your StackOverflow post you had enabled Embedded mode. Do you only see the error in the event log or does the dialog still come up as well? Also can you let me know what your OS Version is? (you can get this by running 'systeminfo' in a cmd window).

nVentis, 2016-11-29:
Hey, the second error I provided in the comment only shows up in the event log. The error that is shown is the one you can see in my original post.

My OS Version is 10.0.14393.

munyirik, 2016-11-29:
Is there any error in the VS output window or does it just hang when you try and deploy?

nVentis, 2016-11-29:
The dialog appears, I click OK, and it disappears. What's really surprising to me is that all this still happens even though I reinstalled Windows through a "vanilla" Microsoft MDT deployment. No configuration / etc. was changed, so I'm wondering why I'm the only one with this error...

munyirik, 2016-11-29:
Can you check if you also have 'Developer mode' set? The setting is in Settings->For developers...
[image]

nVentis, 2016-11-29:
I've checked developer mode and also looked up in the registry if everything is fine (`HKLM/Software/Microsoft/Windows/CurrentVersion/AppModelUnlock`). There, both `AllowAllTrustedApps` and `AllowDevelopmentWithoutDevLicense` are set to `1`.

However, I'm not sure if it's intended behavior that whenever I open the developer options, Windows seems to search for the developer package even though I set the corresponding configuration value to "Developer mode" directly after I installed Windows.

[image]

Here, 

> Das Paket für den Entiwcklermodus wird gesucht...

 would translate to (something like)

> Searching package for developer mode...

munyirik, 2016-11-29:
I think you can ignore the "Searching package..." message. On my desktop the message is that the package is not accessible but I can still deploy.
Just to narrow things down. Can you try deploying a C# or C++ background app? (the templates for them are here - https://marketplace.visualstudio.com/items?itemName=MicrosoftIoT.WindowsIoTCoreProjectTemplates)

nVentis, 2016-11-29:
I get the same error as with Node UWP.

[image]

munyirik, 2016-11-29:
@paulmon any ideas?

paulmon, 2016-11-29:
This error message usually means that the Embedded Mode service is not running.
Try starting it from the services control panel, or by running **sc.exe start embeddedmode**

nVentis, 2016-11-30:
It seems like there is a problem with the service. Whenever I start it, the service seems to stop immediately afterwards.

[image]

[image]

> The service "Embedded mode" was started on "Local Computer" and then stopped.
Some services are stopped automatically if they are not used by other services
or applications.

paulmon, 2016-11-30:
Have you followed these directions to [enable embedded mode](https://developer.microsoft.com/en-us/windows/iot/docs/embeddedmode)? 
Please note that embedded mode is different than developer mode.
You can verify that the embedded mode provisioning package has been added by searching for "Add or remove a provisioning package" in settings.  
Or **Settings>Accounts>Access Work or School>Add or remove a provisioning package**
You should see something like this.
[image]

nVentis, 2016-12-02:
Thank you for your effort, @paulmon and @munyirik !

There seems to be a problem when installing the provisioning package. After installing it, there is no error displayed. However, failing deployments (with the error mentioned above) in Visual Studio already indicate that the package could not be installed.

When restarting, the menu mentioned by @paulmon also shows the following:

[image]

The details sceen:

[image]

The error at the bottom could be translated as

> The configurations (values) were not applied (properly) due to an invalid related property.

Searching the [internet](https://msdn.microsoft.com/windows/hardware/commercialize/customize/mdm/diagnose-mdm-failures-in-windows-10), I found out that related errors should be stored in the Event Logs (Application and service protocols -> Microsoft -> Windows -> DeviceManagement-Enterprise-Diagnostics-Provider).

However, I can only find non-error log entries.

[image]

For the sake of completeness, here follows the provisioning package I created and tried to install.

[allowEmbeddedMode.zip](https://github.com/ms-iot/ntvsiot/files/627283/allowEmbeddedMode.zip)

paulmon, 2016-12-05:
That .ppkg worked when I tested it.
What version and build of windows are you running?
Enterprise, Professional, etc. and build number.
You should be able to see that information by running winver.exe

nVentis, 2016-12-09:
Result of executing winver.exe:

[image]

paulmon, 2016-12-13:
I'm trying to track down how to diagnose the problem.  Thank you for your patience.

vinstce, 2017-04-07:
Hi @nVentis and @paulmon, were you able to find a resolution for this issue? I too facing the same issue and stumbled upon this thread.

m59peacemaker, 2017-05-12:
I'm on Windows 10, 1703 and Visual Studio 2017 and I get the group policy error, too.
I added and removed the AllowEmbedMode .ppkg a bunch of times, as well as adding it when it was already added which produces a different error. The normal error I get is that the policy is not allowed. For no apparent reason, it eventually appears under the account settings as being installed (no error). I still get the above mentioned issue trying to start the EmbeddedMode service. It just stops immediately. But, I noticed that after restarting, the AllowEmbedMode installation error reappears. It says:

```
This package
  Applies policies
    This policy is not allowed
```

nVentis, 2017-10-15:
It seems like the problem has been resolved until now. Haven't tried this for quite a while now, but with Windows 10 15063.632 and Visual Studio 2017, everything seems to work as expected. Note that, in contrast to former Windows versions, only with this version I was able to finally install the PKKG file without getting an error.

Visual Studio will just run the project locally fine. Problem resolved by patience and updating.