ms-iot/ntvsiot #122: Cannot start when using the direct memory mapped driver

patsissons, 2017-03-03:
Testing out the hello world sample, verbatim.

The debugger seems unable to properly attach (breakpoints not hit), but the console is connected as my `console.log()` tests came through. Process exits about a second after creating the http server. As soon as i went back to the inbox driver everything started to work just fine.

NOTE: i was not using any of the new lighting stuff, but perhaps you still need to add the capabilities regardless? If that is the case, might be a good thing to note in the readme file.

munyirik, 2017-03-06:
Hi @patsissons, could you tell me what your IoT Core build number is?

patsissons, 2017-03-06:
Hey sorry, this was on the latest version of IoT Core (i think that is 14393), running on a pi 3.

IoTGirl, 2017-03-06:
Do you know the update value?  The latest is 14393.693 but if you just provisioned the device it could still be 14393.0 without the latest updates.

munyirik, 2017-03-06:
I just tried this. With the capability it works. Without the capability then yes, you will get an exception...
>> The program '[3348] backgroundTaskHost.exe' has exited with code -1073740286 (0xc0000602) 'The process aborted due to a fail fast exception.'.

The node.js lightning sample document mentions the need to add the capabilities ([this link](https://developer.microsoft.com/en-us/windows/iot/samples/helloblinkynodelightning)). 
I can add a note in the sample you used. I'm assuming you were using the steps in [this sample](https://developer.microsoft.com/en-us/windows/iot/samples/helloblinkybackgroundnode)?

patsissons, 2017-03-06:
Yea I had been testing out the new lightning stuff and forgot to revert the driver, so it was a bit unexpected why the sample was not working. I'll note though, my exit code was zero (and debugger wasn't properly attaching), so that was the reason why I didn't immediately think about the driver.

munyirik, 2017-03-06:
Ok. Something in the future that can help with debugging is the --use-logger option. See https://github.com/ms-iot/ntvsiot#console

patsissons, 2017-03-06:
I had done that too, but the log file generated was largely empty. I wasn't getting any context why the process was exiting immediately, just enough context to know it was at least running all of my javascript code before exiting.