ms-iot/ntvsiot #123: uwp.projectNamespace loading class library

nkarakotas, 2017-03-20:
Hi, 

I tried to load a class library using uwp.projectNamespace('MyLib') and it seems to partially load it.
I can see all the classes in object viewer but when I try to access them only the namespace is available. I have some public enums in the library and I cant seem to access them also.
Are we able to load class libraries or only windows runtime component?

Nikolas

IoTGirl, 2017-03-20:
Hi @nkarakotas,

This ntvsiot GitHub starts it's explanation with:

> The NTVS (Node.js Tools for Visual Studio) UWP Extension enables developers to deploy Node.js as a Universal Windows Platform (UWP) application 

Can you confirm that your project is in Node JS?  In looking at [readme.md](https://github.com/ms-iot/ntvsiot/blob/vs2017/README.md) can you explain where the failure happens for you when following these instructions?

Sincerely,
IoTGirl

nkarakotas, 2017-03-20:
Hi,

Yes the project is in Node JS using visual studio project wizard.
In readme here onwards:
##Referencing winmd files You can reference a C# or C++ UWP component in your Node.js UWP app by using the Visual Studio reference manager. 

Example of library loaded:
[image]

munyirik, 2017-03-20:
Hi @nkarakotas only windows runtime components are supported.

closed 2017-04-18 by munyirik