ms-iot/ntvsiot #124: Deleting my "scripts" when deploying

sethreidnz, 2017-03-30:
I am attempting to use the UWP tools for deploying a node server to my Raspbery Pi 3 and it seems to want to replace my scripts section when I deploy. Example is here

[https://github.com/sethreidnz/Node.Iot.Experiment](https://github.com/sethreidnz/Node.Iot.Experiment)

I don't really understand why it would need to do this at all... Any ideas on how we can fix this? Another approach would be to be able to do this in my prebuild

```xml
<PropertyGroup>
  <PreBuildEvent>
      cd Client
      npm run build
  </PreBuildEvent>
</PropertyGroup>
```

But that doesn't seem to work either... I am trying to learn this stuff so I can show it off at user groups etc and get people excited about IoT with JavaScript on Windows 10 but it seems like the tooling just gets in the way.

munyirik, 2017-04-17:
Hi @sethreidnz,
Deployment shouldn't be deleting what's been manually added to scripts so that's a bug. 
If I understand correctly, you just want to run the 'npm run build' command automatically every time you build?
Using prebuild in the project file should work but I think you need to put the command in a \<command\> tag (see https://msdn.microsoft.com/en-us/library/dd293582.aspx)

munyirik, 2017-04-17:
Something else to note is that building packages with native code requires using [these](https://github.com/ms-iot/ntvsiot#installing-npm-packages) instructions. Also the code you're trying to build needs to be UWP compatible.
UWP can be limiting depending on what you're trying to do.
If you want to use 'node.exe' command line you have the option to use [node-chakracore](https://github.com/nodejs/node-chakracore/releases). You would need to copy node and your app files manually to the device so that's the downside. Debugging can be done with [VS Code](https://code.visualstudio.com/docs/nodejs/nodejs-debugging#_remote-debugging) in this case.

sethreidnz, 2017-04-17:
Yes all I'm doing is building my client app (as you would in any other web project) and copying those files into a node application that will serve up the plain old HTML client app.

the issue is that when I run a deploy from Visual Studio it seems it looks at my root `package.json` file and replaces the whole script section so before it is like this:

```json
"scripts": {
    "start": "--debug Server\\server.js",
    "build-client": "cd ./Client && npm run build"
  }
```

After it is like this:

```json
"scripts": {
    "start": "--debug Server\\server.js"
  }
```

Meaning I cannot really add any custom scripts to my npm build...

As to your other comment the Chakra Core method is where I want to go but the details on this and how to do it are hard to find... Where's some good docs on doing the manual deploy and debug with VSCode etc? I want to do some meetups and blog posts about doing pure JS dev on Windows IoT Core...

munyirik, 2017-04-18:
The samples below would be a good starting point:
https://developer.microsoft.com/en-us/windows/iot/samples/helloblinkynode
https://developer.microsoft.com/en-us/windows/iot/samples/memorystatusnodejs

Couple of things to note:
1. You should use the latest release of node-chakracore from [here](https://github.com/nodejs/node-chakracore/releases). The sample docs still point to older versions.
2. The samples also show how to copy files to the device (using file explorer). You can access your device with "\\\\\<device IP address>\\$" so you can use command line as well (we don't have an automatic deploy mechanism yet with VS Code for node-chakracore).
3. To access UWP APIs with node-chakracore you can use [NodeRT](https://github.com/NodeRT/NodeRT)

Let me know if you have any other questions. 
Thanks.