ms-iot/ntvsiot #126: Cannot deploy Basic Node.JS Express 4 Application (Universal Windows) with VS2017

wind-rider, 2017-04-16:
I'm using VS2017 + latest updates. I installed https://github.com/ms-iot/ntvsiot/releases/tag/v1.5.7.

When creating a `Basic Node.JS Express 4 Application (Universal Windows)` and try to deploy it to Windows IoT Core 15063, I get the following errors:

```
------ Build started: Project: Express4UwpApp1, Configuration: Release ARM ------
------ Deploy started: Project: Express4UwpApp1, Configuration: Release ARM ------
DEP6721: Failed to deploy optional package 'Unknown' (B968CC6A-D2C8-4197-88E3-11662042C291). Package is not defined for OneCore.
DEP6721: Failed to deploy optional package 'Unknown' (8CDEABEF-33E1-4A23-A13F-94A49FF36E84). Package is not defined for OneCore.
Updating the layout...
Copying files: Total <1 mb to layout...
Checking whether required frameworks are installed...
Framework: Microsoft.VCLibs.140.00/ARM, app package version 14.0.24123.0 is not currently installed.
Installing missing frameworks...
DEP8000 : Unexpected deployment failure : COMException : error 80080200: App manifest validation error: The app manifest schema must be valid: Reason: /schema/complexType[6][@name = 'CT_Protocol']/all[1]/element[3]
Undeclared XSD element : '{http://schemas.microsoft.com/appx/manifest/foundation/windows10/restrictedcapabilities/3}MigrationProgIds'.

Registering the application to run from layout...
DEP8000 : Unexpected deployment failure : RemoteCommandException : Windows cannot install package Express4UwpApp1_1.0.0.0_arm__heepqs1etyd7r because this package depends on a framework that could not be found. Provide the framework "Microsoft.VCLibs.140.00" published by "CN=Microsoft Corporation, O=Microsoft Corporation, L=Redmond, S=Washington, C=US", with neutral or ARM processor architecture and minimum version 14.0.24123.0, along with this package to install. The frameworks with name "Microsoft.VCLibs.140.00" currently installed are: {Microsoft.VCLibs.140.00_14.0.22929.0_arm__8wekyb3d8bbwe}

0x80073cf3
DEP0700: Registration of the app failed. [0x80004005] Failed to register through new deployment pipeline.
========== Build: 1 succeeded, 0 failed, 0 up-to-date, 0 skipped ==========
========== Deploy: 0 succeeded, 1 failed, 0 skipped ==========
```

munyirik, 2017-04-17:
Hi @wind-rider,
That's a known bug. To workaround the problem, first deploy a C++ UWP app to your device (see image below). You only need to do this one time. The problem is that the Node.js project system doesn't automatically deploy the Microsoft.VCLibs.140.00 framework which it requires.

[image]

wind-rider, 2017-04-25:
@munyirik does it matter which toolset (v140 or v141) is being used?

Deploying the app type you showed doesn't help ;/

munyirik, 2017-04-25:
It shouldn't matter. The template may not be there if you didn't install C++ tools with VS 2017 (not 100% sure about this). Can you try deploying a C# universal windows app? I believe the vc libs will get copied with that type of app as well.

munyirik, 2017-04-25:
Ok, I'll need to look into it. There's also [another issue](https://social.msdn.microsoft.com/Forums/en-US/b072eea2-c0ad-4371-a512-90db64dc6221/express-error-with-require-in-nodejs-app-raspberry-pi3?forum=WindowsIoT) with the Express (using Node.js UWP) which would block you from running the app once you deployed.

In the meantime you can still run the console version node.js on IoT Core using [node-chakracore](https://github.com/nodejs/node-chakracore/releases). 

See a sample here - https://developer.microsoft.com/en-us/windows/iot/samples/helloblinkynode

**Note:**
1. You should use the latest release of node-chakracore from [here](https://github.com/nodejs/node-chakracore/releases). The sample docs still point to older versions.
2. For remote debugging you can use the instructions [here](https://github.com/Microsoft/nodejstools/wiki/Advanced-Debugging#remote-debugging-windows-linux-macos). You can also use VS Code to do [remote debugging](https://code.visualstudio.com/docs/nodejs/nodejs-debugging#_remote-debugging).
3. The samples also show how to copy files to the device (using file explorer). You can access your device with "\\\\\<device IP address>\\$" so you can use command line as well.
4. If you need to access UWP APIs with node-chakracore, you can use [NodeRT](https://github.com/NodeRT/NodeRT)