ms-iot/ntvsiot #127: No Application templates with Visual Studio 2017

binaryangel-noah, 2017-05-14:
Hallo,

i installed the extension on VS2017 and dont see the templates for nodejs uwp. The last releast that has been made is for VS2017 RC. 
Is this Repository still maintained and can we hope for a update to current version of Visual Studio 2017 ?

IoTGirl, 2017-05-16:
Hi @binaryangel-noah ,

Have you reviewed the already open issue https://github.com/ms-iot/ntvsiot/issues/126 for steps to use with VS2017?

Sincerely, IoTGirl

milt0r, 2017-05-16:
I had the same issue. Install ntvsiot from here: https://github.com/ms-iot/ntvsiot/releases/tag/v1.5.7

robintel, 2017-06-28:
Having installed the 1.5.7 version, and allowing for NodeJS development in Visual Studio 2017, I still don't see the template.