ms-iot/ntvsiot #128: Unable to query Azure DB from Windows 10 IoT Core

munyirik, 2017-06-14:
Version: v5.10.1-0
Platform: Windows 10 IoT Core
Subsystem: node-mssql , tedious

I'm unable to query Azure MS SQL DB from Windows 10 IoT Core (Raspberry Pi). Will this ever be supported or it already should be?

I've created sample so we can reproduce issue:
https://github.com/matejv1/nodejs.universal-windows-app.connect-to-azure

It works on Windows 10 - if I just run "node server.js". When debugging or deploying to Raspberry Pi, the app just crashes.

Hopefully someone will give some feedback.

I've posted an issue to node-mssql repository as well. Might be tedious the issue.

Thanks.

munyirik, 2017-06-14:
cc @paulmon @msalehmsft

digitalinfinity, 2017-06-14:
cc @matejv1 in case you're still interested in this issue